veripool/verilog-mode #1820: Reversion of bug 1257: Indentation within generate construct after always block is wrong if generate/endgenerate omitted

dbhopper, 2023-01-17:
I was the original reporter of what was ported into Github as [issue 1257](https://github.com/veripool/verilog-mode/issues/1257), which was fixed by Punzik on 9/14/2021.  I appreciate the fix, BTW!  I didn't realize until recently the issue had been resolved. 

However, it appears the current git rev has a behavioral reversion on this issue.  It once again doesn't properly indent such code in generate blocks as 1257 posters detail. It looks like some of the patched code was totally rewritten in a [Oct 16 commit by Gonzalo Larumbe](https://github.com/veripool/verilog-mode/commit/ce34ef91433948c5e7aff3a70f93b09ae5900929). It may be the source of the bug's reappearance: 

commit ce34ef91433948c5e7aff3a70f93b09ae5900929
Author:     Gonzalo Larumbe <gonzalomlarumbe@gmail.com>
AuthorDate: Sun Oct 2 17:33:59 2022 +0200
Commit:     Gonzalo Larumbe <51021955+gmlarumbe@users.noreply.github.com>
CommitDate: Sun Oct 16 12:49:32 2022 +0200

    Fix bug in verilog-in-generate-region-p

Would it be possible to re-run the test cases from issue 1257, including my original example, and either fix the current verilog-in-generate-region-p code, or revert to the pre-ce34ef91 code here?  Thanks!

Dan

wsnyder, 2023-01-17:
@gmlarumbe, might you take a look?

gmlarumbe, 2023-01-18:
Hi @dbhopper ,

Thanks for reporting the issue.

Changes of ce34ef9 are part of PR #1815. There is a snippet that shows some of the problems with `verilog-in-generate-region-p`.

For some reason I added the `generate/endgenerate` keywords to the tests (maybe while working on a different issue) and that hid indentation errors. My bad, sorry for the confusion.

I have reverted the changes from ce34ef9 and updated the tests in 2034ab7. Could you please check if it works as expected now?

Thanks!