veripool/verilog-mode #1821: how to connect port with different wire

zhanjf, 2023-02-04:
wire [2:0] a;
wire [3:0] b;

if I want to combine a and b like {5'b0, a[2:0], 4' b0, b[3:0]} and connect one 16 bits width port ?
how can I calculate the filled zero bit width? 

Thanks.

zhanjf, 2023-02-04:
It's possible to get bit width of wire signal like vl-width for port using LISP?

wsnyder, 2023-02-04:
I'd suggest instead use verilog parameters (for the "2" and "3") then math in Verilog.

In Verilog-Mode it would be something like this, you'd call it as a function in the AUTOINST, it'll require some debug:

```
(defun my-port-width (sig)
  (let* ((modi (verilog-modi-current))
         (decls (verilog-modi-get-decls modi))
         (port-info (assoc sig (verilog-decls-get-signals decls))))
    (when port-info (verilog-sig-bits port-info))))
```

closed 2023-02-04 by wsnyder