veripool/verilog-mode #1822: Syntax highlighting confused by certain type of comment block

jwcollins, 2023-02-07:
Hi.  I'm running xemacs21 with the latest verilog-mode as of commit 5fe663db, on Ubuntu 22.04 Linux.  (This has been a problem since at least commit 2034ab72.)

This type of comment block confuses the syntax highlighting.  (It never used to be a problem with older versions of verilog-mode.)
/*******************************************************************************
 * tb_clk.sv
 *
 * For testbench purposes, create a square wave clock signal.
 ******************************************************************************/

When I first open such a file, it's correctly highlighted in blue as a comment.  But if I change the contents of this comment block, the text reverts to color black, as though it's no longer recognized as a comment.

[image]
[image]
[image]

Thanks in advance for any help with this!  --jwc

wsnyder, 2023-02-08:
If you'd like to debug it, I certainly would take the patch, but otherwise I don't spend any effort on xemacs, sorry.

May I ask why you are still using xemacs? It hasn't been updated in 14 years, I was planning on removing support for it.

gmlarumbe, 2023-02-08:
Hi @jwcollins ,

I have tried to reproduce the issue with the snippet you provided without success both in GNU Emacs 29 and XEmacs 21.5. Could you provide the rest of the code? Or if not possible, another whole snippet that also has the same issue?

I also reviewed the changes between 2034ab7 and 5fe663d and it seems they do not include anything related to syntax highlighting. I can give it another try in case anything between these commits had something to do, but as Wilson pointed out it's not really worth spending much time in debugging old versions.

jwcollins, 2023-02-08:
Hi @wsnyder.  Regrettably, I'm not a lisp programmer.  Probably the best I can do is to find what commit initially created this issue.

As to why Xemacs?  Inertia, I guess.  I've developed code in multiple languages using Xemacs since my grad school days in the mid 90's.  If I'm forced to, I'll migrate to Gnu Emacs.  --jwc

jwcollins, 2023-02-09:
Hi @gmlarumbe.  I wasn't claiming that this issue was created between commits 2034ab7 and 5fe663d.  Rather, I was saying that the issue exists at both of those commits.  Thus, the problem originated before 2034ab7.

I'm using XEmacs 21.4, as packaged with Ubuntu 22.04.  The 21.4 branch is the stable branch.  21.5 is beta test.

[image]

I'm attaching a small source file (in the zip) 
[verilog-mode-files.zip](https://github.com/veripool/verilog-mode/files/10691173/verilog-mode-files.zip)
that exhibits this behavior for me.  I see this misbehavior in the title block and block quotes only, and not in ordinary "/* comment */" comment lines.

I'm also attaching my ~/.xemacs/custom.el and init.el files, though I don't think I'm doing anything weird in there.  Finally, I'm attaching an .el file that Ubuntu seems to load.  Hope any of this is helpful.  Thanks.  --jwc

gmlarumbe, 2023-02-09:
I could reproduce it with XEmacs 21.4 and the files you provided, but still works fine under XEmacs 21.5.34 

@jwcollins , could you also try 21.5 and see if the error is still present?

jwcollins, 2023-02-09:
Hi @gmlarumbe.  Thanks for looking into this.  I'm glad to hear that this issue is reproducible.  I can attempt to build XEmacs 21.5, but it may take me a few days to get to it at work here.  The last time I tried building 21.5 (about 2 years ago) on an older (18.04) Ubuntu install, it was quite painful, and didn't produce a stable XEmacs build.