veripool/verilog-mode #1823: What magic sauce is required to indent typedef's correctly?

shareefj, 2023-02-21:
I'm struggling with indentation again and feel like I've been here before but haven't managed to find anything that works.  All the examples and similar issues all point at using `verilog-typedef-regexp` but I haven't managed to get it to work yet.  I do recall that on previous issues where this didn't work, I realised I had to set it in the child module for AUTO's to work, but in this case I've tried pretty much every permutation and it still fails.

As an example, how do I get `my_special_t` to be indented correctly?

```systemverilog
package test_pkg;
  typedef enum logic
    {
     StateOne = 1'b0,
     StateTwo = 1'b1
     } my_special_t;
endpackage

module test
  import test_pkg::*;
  #(
    parameter int TestParam = 1
    )(
      input logic clk,
      input logic rst_n,
      input       my_special_t special
      );
endmodule

// Local Variables:
// verilog-library-directories:(".")
// verilog-typedef-regexp:"_t$"
// End:

```

wsnyder, 2023-02-22:
There's also verilog-align-typedef-regexp, but it doesn't seem to help this.
Another possibility (what I use) is 

// verilog-auto-lineup: nil

Then manually delete indent once as this won't do it for you.

Fixes for this would be welcome.

gmlarumbe, 2023-02-24:
Hi @shareefj ,

@wsnyder might help me out here, but I think that the variable `verilog-typedef-regexp` is only used for AUTOs.

For indentation/alignment you have two options:
- Set the value of `verilog-align-typedef-regexp` to the Elisp regexp of your typedefs. E.g:
```elisp
;; For typedefs that end with *_t:
(setq verilog-align-typedef-regexp (concat "\\<" verilog-identifier-re "_\\(t\\)\\>"))
;; For typedefs that end with *_t, *_if or *_vif:
(setq verilog-align-typedef-regexp (concat "\\<" verilog-identifier-re "_\\(t\\|if\\|vif\\)\\>"))
```
-  Set the value of `verilog-align-typedef-words` to a list of strings that should be considered as typedefs. E.g:
```elisp
;; For this issue snippet:
(setq verilog-align-typedef-words '("my_special_t"))
;; Multiple typedefs:
(setq verilog-align-typedef-words '("my_typedef_1" "my_typedef_2" "my_typedef_3"))
```
Running `verilog-pretty-declarations` on the line `input my_special_t special` should align it correctly. However, the second method is not recommended if you need to add many words since it will slow down your Emacs session on files under `verilog-mode`.

If you want to also fontify/syntax highlight the typedefs set by previous two variables you can have a look at the [verilog-ext](https://github.com/gmlarumbe/verilog-ext) package.