veripool/verilog-mode #1825: 2-D array in AUTO_TEMPLATE getting wrong subscript if [][] is used

Grmarder, 2023-03-03:
Please look in to the indexes of array_out , instead of being [0:7][0:15] it is [0:15][0:15], it happens only  when the port appears in the template 
The workaround is to use : @"vl-mbits"[]  

module twoD_array 
(
 input [0:7] [15:0] array_in1,
 input [0:7] [15:0] array_in2, 
 output [0:7] [15:0] array_out
)

endmodule

module new_one 
  (
   /*AUTOINPUT*/
   // Beginning of automatic inputs (from unused autoinst inputs)
   input [0:7] [15:0]	array_in1,		// To twoD_array_inst of twoD_array.v
   input [0:7] [15:0]	array_in2,		// To twoD_array_inst of twoD_array.v
   // End of automatics
   /*AUTOOUTPUT*/
   // Beginning of automatic outputs (from unused autoinst outputs)
   output [15:0] [15:0]	array_out		// From twoD_array_inst of twoD_array.v
   // End of automatics
   );
   /* twoD_array AUTO_TEMPLATE
    (
    .\(.*in1.*\)			(\1@"vl-mbits"[]),
    .\(.*out.*\)		    (\1[][]),
    );
    */
    twoD_array twoD_array_inst
     (/*AUTOINST*/
      // Outputs
      .array_out			(array_out[15:0][15:0]), // Templated
      // Inputs
      .array_in1			(array_in1[0:7][15:0]),	 // Templated
      .array_in2			(array_in2/*[0:7][15:0]*/));

endmodule

wsnyder, 2023-03-03:
When I use the latest verilog-mode it appears correct, please check that.

Grmarder, 2023-03-04:
You are right !!! With newer version there is no such issue.
Thanks a lot for the speed of light response :)