veripool/verilog-mode #1829: "always @(* )"  will trigger Error message '%Error:Search failed: "*)"'

BlingBlingKing, 2023-06-01:
dut.sv
``` verilog
module dut (/*autoarg*/);
input clk;
input rst_n;
output logic d;

always @(* )
begin
  d = 1'b0;
end
```

Then we run the follow command
```
emacs -batch -l ./verilog-mode.el dut.sv -f verilog-batch-auto
```

The error message shows:
```
%Error: Search failed: "*)"
```
If we replace "@(* )" with "@(*)" or "@( * )", the error message won't show again.

gmlarumbe, 2023-06-02:
Hi @BlingBlingKing ,

Thanks for reporting the issue. The following patch to `verilog-read-decls` seems to fix it:

```diff
diff --git a/verilog-mode.el b/verilog-mode.el
index 37b09b5..b07f85c 100644
--- a/verilog-mode.el
+++ b/verilog-mode.el
@@ -9322,7 +9322,7 @@ Return an array of [outputs inouts inputs wire reg assign const gparam intf]."
 	 ((looking-at "(\\*")
 	  ;; To advance past either "(*)" or "(* ... *)" don't forward past first *
 	  (forward-char 1)
-	  (or (search-forward "*)")
+	  (or (re-search-forward "*?\\s-*)" nil t)
 	      (error "%s: Unmatched (* *), at char %d" (verilog-point-text) (point))))
 	 ((eq ?\" (following-char))
           (or (re-search-forward "[^\\]\"" nil t)  ; don't forward-char first, since we look for a non backslash first
```

Even though tests for Emacs keep passing (could not test for XEmacs though), @wsnyder might know better what the solution is here since I do not know much about the AUTO engine.

wsnyder, 2023-06-06:
It needs to be slightly different, as we need to search for *) specifically to end attributes.

Anyhow thanks for zeroing in in where to fix it. Fixed in master.

closed 2023-06-06 by wsnyder