veripool/verilog-mode #1831: Issue when using `\` before `"`

LapinFou, 2023-06-21:
Hi folks,

I encountered an issue when using AUTO features with the following Verilog code.
```
Updating AUTOs...
verilog-modi-current: Search failed: "\\<\\(end\\(?:c\\(?:lass\\|on\\(?:fig\\|nectmodule\\)\\)\\|interface\\|module\\|p\\(?:ackage\\|r\\(?:imitive\\|ogram\\)\\)\\)\\)\\>"
```
After some investigation, I found the problematic line (see below).
```
   task tbPassed;
      begin
           begin
              $display("\n                  xx");
              $display("                xxx    _____         _____ _____");
              $display("              xxxx    |  __ \\ /\\    / ____/ ____|");
              $display("            xxxxx     | |__) /  \\  | (___| (___");
              $display("  xxx     xxxxxx      |  ___/ /\\ \\  \\___ \\\\___ \\"); // Generate error -> "verilog-modi-current: Search failed [...]"
              $display("  xxxx  xxxxxx        | |  / ____ \\ ____) |___) |");
              $display("    xxxxxxxx          |_| /_/    \\_\\_____/_____/");
              $display("      xxxx");
          end
      end
   endtask
```
If I add a space before the final `"` (`\\ ")` instead of `\\")`), then it does work.

wsnyder, 2023-07-25:
Thanks for the report.  I looked at this and the possible fix would be very slow, so I think until a better fix gets more thought it will need to be worked around as you did.

acr4, 2023-09-09:
PR #1840 begins to think about how to solve this. Like @wsnyder said, it's more complex than it looks on the surface. Here's a quick test-case, but by no means an exhaustive one. It just builds on @LapinFou's original text block.

```
module m;

   always @(/*AS*/) begin
      if(x)
        y = z;
   end

   task tbPassed;
      begin
         begin
            $display("\n                  xx");
            $display("                xxx    _____         _____ _____");
            $display("              xxxx    |  __ \\ /\\    / ____/ ____|");
            $display("            xxxxx     | |__) /  \\  | (___| (___");
            $display("  xxx     xxxxxx      |  ___/ /\\ \\  \\___ \\\\___ \\");
            $display("  xxxx  xxxxxx        | |  / ____ \\ ____) |___) |");
            $display("    xxxxxxxx          |_| /_/    \\_\\_____/_____/");
            $display("      xxxx");
         end
      end
   endtask

endmodule // m
```

LapinFou, 2023-09-09:
Thanks a lot for looking to fix this issue. 👍