veripool/verilog-mode #1835: How to make verilog mode not add output for internal signals?

bharanr, 2023-08-08:
I did AUTOINST for a module instance. The output from the module is used in another always_comb block.
I manually did logic declaration for that signal but verilog mode still adds it the port declaration.
Is that a way to avoid this?

Example:
module foo(
 (
   /*AUTOOUTPUT*/
   /*AUTOINPUT*/
 );

logic a,b;
/* AUTOWIRE*/

/*bar AUTO_TEMPLATE (
 ); */                                                                                                                                                                                        
bar bar0 (
  /*AUTOINST*/
);

endmodule

// Local Variables:
// verilog-library-directories:(".")
// End:

Verilog-mode Auto output
module foo(
 (
   /*AUTOOUTPUT*/
   // Beginning of automatic outputs (from unused autoinst outputs)
   output logic   a,      // From bar0 of bar.v
   output logic   b,      // From bar0 of bar.v
   // End of automatics
   /*AUTOINPUT*/
   // Beginning of automatic inputs (from unused autoinst inputs)
   input    c     // To bar0 of bar.v
   // End of automatics                                                                                                                                                                       
 );

logic a,b;
/* AUTOWIRE*/

/*bar AUTO_TEMPLATE (
 ); */
bar bar0 (
  /*AUTOINST*/
    // Outputs
    .a        (a),
    .b        (b),
    // Inputs
    .c        (c));

endmodule

// Local Variables:
// verilog-library-directories:(".")
// End:

Expectation:
module foo(
 (
   /*AUTOOUTPUT*/
   /*AUTOINPUT*/
   // Beginning of automatic inputs (from unused autoinst inputs)
   input    c     // To bar0 of bar.v
   // End of automatics                                                                                                                                                                       
 );

logic a,b;
/* AUTOWIRE*/

/*bar AUTO_TEMPLATE (
 ); */
bar bar0 (
  /*AUTOINST*/
    // Outputs
    .a        (a),
    .b        (b),
    // Inputs
    .c        (c));

endmodule

// Local Variables:
// verilog-library-directories:(".")
// End:

Emacs  : GNU Emacs 24.3.1 (x86_64-redhat-linux-gnu, GTK+ Version 3.22.30)
 of 2020-04-03 on x86-01.bsys.centos.org
Package: verilog-mode vUNKNOWN_VER-UNKNOWN_REV-vpo

current state:
==============
(setq
 verilog-active-low-regexp nil
 verilog-after-save-font-hook nil
 verilog-align-assign-expr nil
 verilog-align-comment-distance 1
 verilog-align-decl-expr-comments t
 verilog-align-ifelse nil
 verilog-align-typedef-regexp nil
 verilog-align-typedef-words nil
 verilog-assignment-delay ""
 verilog-auto-arg-sort nil
 verilog-auto-declare-nettype nil
 verilog-auto-delete-trailing-whitespace nil
 verilog-auto-endcomments t
 verilog-auto-hook nil
 verilog-auto-ignore-concat nil
 verilog-auto-indent-on-newline t
 verilog-auto-inout-ignore-regexp nil
 verilog-auto-input-ignore-regexp nil
 verilog-auto-inst-column 40
 verilog-auto-inst-dot-name nil
 verilog-auto-inst-interfaced-ports nil
 verilog-auto-inst-param-value nil
 verilog-auto-inst-sort nil
 verilog-auto-inst-template-numbers nil
 verilog-auto-inst-vector t
 verilog-auto-lineup 'declarations
 verilog-auto-newline t
 verilog-auto-output-ignore-regexp nil
 verilog-auto-read-includes nil
 verilog-auto-reset-blocking-in-non t
 verilog-auto-reset-widths t
 verilog-auto-save-policy nil
 verilog-auto-sense-defines-constant nil
 verilog-auto-sense-include-inputs nil
 verilog-auto-star-expand t
 verilog-auto-star-save nil
 verilog-auto-template-warn-unused nil
 verilog-auto-tieoff-declaration "wire"
 verilog-auto-tieoff-ignore-regexp nil
 verilog-auto-unused-ignore-regexp nil
 verilog-auto-wire-type nil
 verilog-before-auto-hook nil
 verilog-before-delete-auto-hook nil
 verilog-before-getopt-flags-hook nil
 verilog-before-save-font-hook nil
 verilog-cache-enabled t
 verilog-case-fold t
 verilog-case-indent 2
 verilog-cexp-indent 2
 verilog-compiler "echo 'No verilog-compiler set, see \"M-x describe-variable verilog-compiler\"'"
 verilog-coverage "echo 'No verilog-coverage set, see \"M-x describe-variable verilog-coverage\"'"
 verilog-delete-auto-hook nil
 verilog-fontify-variables t
 verilog-getopt-flags-hook nil
 verilog-highlight-grouping-keywords nil
 verilog-highlight-includes t
 verilog-highlight-modules nil
 verilog-highlight-translate-off nil
 verilog-indent-begin-after-if t
 verilog-indent-class-inside-pkg t
 verilog-indent-declaration-macros nil
 verilog-indent-ignore-multiline-defines t
 verilog-indent-ignore-regexp nil
 verilog-indent-level 3
 verilog-indent-level-behavioral 3
 verilog-indent-level-declaration 3
 verilog-indent-level-directive 1
 verilog-indent-level-module 3
 verilog-indent-lists t
 verilog-library-directories '(".")
 verilog-library-extensions '(".v" ".va" ".sv")
 verilog-library-files nil
 verilog-library-flags '("")
 verilog-linter "echo 'No verilog-linter set, see \"M-x describe-variable verilog-linter\"'"
 verilog-minimum-comment-distance 10
 verilog-mode-hook '(verilog-set-compile-command)
 verilog-mode-release-emacs nil
 verilog-mode-version "UNKNOWN_VER-UNKNOWN_REV-vpo"
 verilog-preprocessor "verilator -E __FLAGS__ __FILE__"
 verilog-simulator "echo 'No verilog-simulator set, see \"M-x describe-variable verilog-simulator\"'"
 verilog-tab-always-indent t
 verilog-tab-to-comment nil
 verilog-typedef-regexp nil
 verilog-warn-fatal nil
 )

my2817, 2023-08-10:
reper to help doc of ```verilog-auto-output-ignore-regexp```

wsnyder, 2023-08-10:
To find ```verilog-auto-output-ignore-regexp``` use emacs help, or search for it in the docs here https://veripool.org/verilog-mode/help/

closed 2023-08-10 by wsnyder

bharanr, 2023-08-17:
Thanks for the help @my2817 and @wsnyder