veripool/verilog-mode #1836: Fix indention for restrict_property_statement

pbing, 2023-08-27:
The restrict_property_statement leads to wrong indention:
```verilog
module tb1;
   a: restrict property (1);
      b: assume property (1);
      c: assume property (1);
      endmodule
```

It should be:
```verilog
module tb1;
   a: restrict property (1);
   b: assume property (1);
   c: assume property (1);
endmodule
```

wsnyder, 2023-08-27:
Perhaps you could contribute a pull request to fix this?

acr4, 2023-08-27:
See PR [#1838](https://github.com/veripool/verilog-mode/pull/1838) for a proposed solution.

closed 2023-08-27 by wsnyder

kzhang70, 2023-10-23:
The fix does NOT work if property is not in the same line as the label.

module tb1;
   a:
     restrict property (1);
      b:
        assume property (1);
         c:
           assume property (1);
            endmodule