veripool/verilog-mode #1841: How to align the parenthesis generated by autoinst?

zheBytedance, 2023-09-13:
Hey, thank you for your effort in bringing such a great tool.

But I have a question related to the auto-inst, I find the code generated is not aligned by parenthesis. I must realign manually every time and once I rerun the `!emacs -l ~/bin/verilog-mode.el --batch % -f verilog-batch-auto<CR> `  in vim. It just go back to the following unaligned parenthesis. How do I solve this?

```verilog
    dut #(
        /*autoinstparam*/
       // Parameters
       .NUM_ELEMENTS        (NUM_ELEMENTS),
       .ID_WIDTH            (ID_WIDTH),
       .PRIO_WIDTH          (PRIO_WIDTH))
        dut_inst
    ( /*autoinst*/
     // Outputs
     .push_ready_o          (push_ready[gvi]),   // Templated
     .pop_valid_o           (pop_valid[gvi]),    // Templated
     .pop_priority_o            (pop_priority[gvi]),     // Templated
     .pop_id_o              (pop_id[gvi]),       // Templated
     // Inputs
     .clk_i             (clk_i),
     .reset_i               (reset_i),
     .push_valid_i          (push_valid[gvi]),   // Templated
     .push_priority_i           (push_priority[gvi]),    // Templated
     .push_id_i             (push_id[gvi]),      // Templated
     .pop_ready_i           (pop_ready[gvi]),    // Templated
     .clear_all_i           (clear_all_i));
```

wsnyder, 2023-09-13:
Sorry, there isn't any option for indenting differently the autogenerated blocks, and this is unlikely to get added, but perhaps you could reindent everything?

```
emacs -l ~/bin/verilog-mode.el --batch % -f verilog-batch-auto -v verilog-batch-indent
```

closed 2023-09-13 by wsnyder

zheBytedance, 2023-09-21:
My colleague found a good solution. Add the following at the end of you file.
// Local Variables:
// indent-tabs-mode: nil
// tab-width: 4
// End: