veripool/verilog-mode #1842: Remove the modport of the interface from AUTOINST

szkarn, 2023-09-20:
Hi,
i wonder how to remove or do not generate the modport of the interface when using AUTOINST

like removing the [.master] of [.svi_modport  (svi_modport.master),] in below codes:

```
interface svi;
   logic       enable;
   modport master (input enable);
endinterface

module InstModule
  (input      clk,
   svi.master svi_modport,
   svi        svi_nomodport);
endmodule

module top;
   InstModule instName
     (/*AUTOINST*/
      // Interfaces
      .svi_modport         (svi_modport.master),
      .svi_nomodport       (svi_nomodport),
      // Inputs
      .clk                 (clk));
endmodule
```

wsnyder, 2023-09-20:
Add it yourself before the AUTOINST.

```
module top;
   InstModule instName
     (
      // Interfaces
      .svi_modport         (svi_modport),
      /*AUTOINST*/
      // Interfaces
      .svi_nomodport       (svi_nomodport),
      // Inputs
      .clk                 (clk));
endmodule
```

closed 2023-09-20 by wsnyder

wsnyder, 2023-09-20:
If you have a modport in the cell you are instancing it should also be where it's instantiated, it's better for readability.  It's also strange if you are using modports for one module but not another, that's legal, just unusual.  I'm still not understanding what is breaking for you with what verilog-mode does by including the modport, you talk about it affecting another usage but that's not shown and would be a different instance that should still properly connect using AUTOs.

szkarn, 2023-09-20:
Thanks for your response. An example shown below:
```
interface svi;
   logic       enable;
   modport master (input enable);
   modport slave (output enable);
endinterface

module MasterModule
  (input      clk,
   svi.master svi_modport);
endmodule

module SlaveModule
  (input      clk,
   svi.slave svi_modport);
endmodule

module top;
   ...
   InstModule MasterModule
     (/*AUTOINST*/
      // Interfaces
      .svi_modport         (svi_modport.master),
      // Inputs
      .clk                 (clk));
   InstModule SlaveModule
     (/*AUTOINST*/
      // Interfaces
      .svi_modport         (svi_modport.slave),
      // Inputs
      .clk                 (clk));
endmodule
```
it seems like I need an extra assign svi_modport.master = svi_modport.slave,  which would not generated by "AUTO".
cause I'm not familiar with systemverilog, could you give me some suggestions about it?

wsnyder, 2023-09-23:
You need to instantiate the interface in the upper module.  Then it can connect as svi_modport, svi_modport.master, or svi_modport.slave.

```
module t;
   svi svi_modport;
   InstModule instName
     (/*AUTOINST*/
      // Interfaces
      .svi_modport                      (svi_modport.master));
endmodule

```