veripool/verilog-mode #1844: Wrong auto extremes computation when signal is connected to input and output

boorbajones, 2023-10-05:
Howdy!

Whilst playing with AUTO for a demo I found a curious error, which is a bit "on the edge" in terms of usage, but still not impossible.

It originated from an example of 2nd level metaprogramming, but it holds also if you reduce the matter to a 1st level (e.g.: using the AUTO_TEMPLATE at first round, and then removing the "wrapping" AUTOINSERTLISP code.

module c(/*AUTOARG*/);
  output yy;
  input  ww;

  xx[0] = ww;
  /*AUTOINSERTLISP(insert (format "  yy = xx[%d];\n" atop))*/
  /*AUTOINPUT*/
  /*AUTOOUTPUT*/
  /*AUTOWIRE*/
  /*AUTO_LISP(setq atop 5)*/
  /*d AUTO_TEMPLATE (
   .y(xx[@"(1+ @)"]),
   .x(xx[@]),
   )
   */
  /*AUTOINSERTLISP(dotimes (a atop) (insert (format "d d%d(/*AUTOINST\*\/);\n" a)))*/
endmodule

module d(/*AUTOARG*/);
  input  bit x;
  output bit y;

  assign y=!x;
endmodule : d

If you try it you'll see that xx is wrongly sized. The xx[0] input of d0 is not accounted for.

I tried to debug (verilog-auto-wire) calls, but even though I got to the point where I was seeing the two identified lists for inputs and outputs of instantiated submodules, I did not manage to converge to a resolving patch.

This all affected version 2023.08.27.008533849

Cheers!

  Andrea.

wsnyder, 2023-10-05:
AUTOWIRE only declares based on outputs of an AUTOINST.  [0] is not such.  You'll need to declare it manually.

wsnyder, 2023-10-05:
P.S. To do that

    bit [XX_MSB:0] xx;
    /*AUTOINSERTLISP(insert (format "  localparam XX_MSB = %d;\n" atop))*/

closed 2023-10-05 by wsnyder

boorbajones, 2023-10-06:
Inspired by your answer I think I found a patch for verilog-auto-wire that works on my example; it also passes the test suite (I had to hack it a bit as my emacs setup saves file without tabs and trailing spaces (so I had to change the diff step into a diff -w.

How can I proceed to submit the change? Is a pull request OK?

wsnyder, 2023-10-06:
Great!  Yes, a pull request is appreciated, please make sure to add a test (and test_ok as golden) as part of the pull, thanks!