veripool/verilog-mode #1846: How to modify the distance/space/indent of the declaration in the AUTOINPUT/AUTOOUTPUT ？

szkarn, 2023-11-15:
Hello,  I have a question about using the AUTOINPUT/AUTOOUTPUT to declare the input/output ports, which have a long parameter name in the bit-width declaration. It exceeds the default space between the "input" and "port_name". This happens a lot that affects the alignment for reading.

For example:
```
module top(
/*AUTOINPUT*/
Input             a1,
Input  [1:0]      a2,
Input  [para_mmmm-1:0]  a3,
Input  [para_nn-1:0]  a4);
...
endmodule
```

Is there a method to modify the default space/indent in the declaration?

Thank you for any tips.

wsnyder, 2023-11-16:
AUTOINDENT shouldn't ever indent like that by itself, I suspect you are using the verilog-mode indent as a second later step. To control that indent you likely want

(setq-default verilog-auto-lineup nil)

Closing as answered but feel free to follow up here.

closed 2023-11-16 by wsnyder

szkarn, 2023-11-16:
It is just because there is a very long parameter name in the sub-module. The length exceed default space of the declaration. I wonder if there is a setting to control the indent space between the "input" and "port_name" in the AUTOINPUT. 

By the way, what is AUTOINDENT?

wsnyder, 2023-11-16:
Can you attach an example of what you get, and what you want to get?

szkarn, 2023-11-16:
what I get:
```
module top(
/*AUTOINPUT*/
Input             a1,
Input  [1:0]      a2,
Input  [para_mmmm-1:0]  a3,
Input  [para_nn-1:0]  a4);
...
endmodule
```

I want to get:
```
module top(
/*AUTOINPUT*/
Input                   a1,
Input  [1:0]            a2,
Input  [para_mmmm-1:0]  a3,
Input  [para_nn-1:0]    a4);
...
endmodule
```

I think the default space is fixed, can it be modified by setting?

wsnyder, 2023-11-16:
Ah. You could reindent the buffer afterwards, e.g. (setq verilog-auto-lineup 'declarations) then verilog-indent-buffer.  Alternatively please consider contributing a pull request to make this a variable, along with a test.

szkarn, 2023-11-17:
Thanks, but I run the verilog-mode in VIM, the buffer should be modified in EMACS?

wsnyder, 2023-11-17:
Huh? Verilog mode only exists in emacs.  If you are calling it otherwise it must be by ascript so fix that script.

szkarn, 2023-11-17:
Yeap, the vim call the verilog-mode.el by a macro plugin, as described in the FAQ: 
https://github.com/veripool/verilog-mode/blob/master/FAQ.rst#can-i-run-verilog-mode-from-vim

It means I should modify the VIM macro? Is there a easier way to do this, like modifying the value in verilog-mode.el directly.

szkarn, 2023-11-20:
which field should I modify in the verilog-mode.el ?
I try to modify the "indent" related variables in the User customization, but not work.

wsnyder, 2023-11-20:
As above "alternatively please consider contributing a pull request to make this a variable, along with a test", there is no current variable for this.

szkarn, 2023-11-21:
Thanks for you patient reply.