veripool/verilog-mode #1847: Using AUTOs with parameterized type

cswmeta, 2023-11-27:
I have tried AUTO expansion with parameterized types and it works with the example code given in https://github.com/veripool/verilog-mode/issues/1061.

I have a case where parameterized types work properly when the child module is declared in the same file as its parent (like in issue 1061). However, if I move the child module to its own file, then the parameter type is not correctly reflected in the AUTOINPUT/AUTOOUTPUT declarations, and the comments in the module instance now seem to think the parameterized port types are Interfaces instead of ports.

Any help to point out what I'm doing wrong is most appreciated.

Here is the example file where it works correctly:

```
module test
(
  /*AUTOINPUT*/
  // Beginning of automatic inputs (from unused autoinst inputs)
  input logic           clk,                    // To i_fifo of fifo.v, ...
  input logic [7:0]     din,                    // To i_fifo of fifo.v
  input logic           rst_n,                  // To i_fifo of fifo.v, ...
  // End of automatics
  /*AUTOOUTPUT*/
  // Beginning of automatic outputs (from unused autoinst outputs)
  output logic [7:0]    dout                    // From i_fifo of fifo.v
  // End of automatics
);

  /*AUTOLOGIC*/
  // Beginning of automatic wires (for undeclared instantiated-module outputs)
  logic [7:0]           d;                      // From i_fifo of fifo.v
  // End of automatics

  /* fifo  AUTO_TEMPLATE (
    .dout (d),
  ); */

  fifo #
    (.DEPTH   (24),
     .T_DATA  (logic [7:0]))
  i_fifo
    (/*AUTOINST*/
     // Outputs
     .dout                              (d),                     // Templated
     // Inputs
     .clk                               (clk),
     .rst_n                             (rst_n),
     .din                               (din));

  /* fifo  AUTO_TEMPLATE (
    .din  (d),
  ); */

  fifo #
    (.DEPTH   (24),
     .T_DATA  (logic [7:0]))
  i_fifo
    (/*AUTOINST*/
     // Outputs
     .dout                              (dout),
     // Inputs
     .clk                               (clk),
     .rst_n                             (rst_n),
     .din                               (d));                    // Templated

endmodule

module fifo #(
  parameter DEPTH = 2,
  parameter type T_DATA = logic [7:0]
) (
  input  logic            clk,
  input  logic            rst_n,
  input  T_DATA           din,
  output T_DATA           dout
);

endmodule

// Local Variables:
// verilog-typedef-regexp: "^[tT]_"
// verilog-auto-inst-param-value:t
// End:
```

Now, if I remove the declaration for the fifo module from the same file as the test module and place it into its own file (called fifo.sv), then you can see the expanded results here. Notice no din or dout declarations, nothing under AUTOLOGIC, and the comments in the fifo instances related to din and dout use "Interfaces" whereas before they were under "Inputs" and "Outputs"

```
module test
(
  /*AUTOINPUT*/
  // Beginning of automatic inputs (from unused autoinst inputs)
  input logic           clk,                    // To i_fifo of fifo.v, ...
  input logic           rst_n                   // To i_fifo of fifo.v, ...
  // End of automatics
  /*AUTOOUTPUT*/
);

  /*AUTOLOGIC*/

  /* fifo  AUTO_TEMPLATE (
    .dout (d),
  ); */

  fifo #
    (.DEPTH   (24),
     .T_DATA  (logic [7:0]))
  i_fifo
    (/*AUTOINST*/
     // Interfaces
     .din                               (din),
     .dout                              (d),                     // Templated
     // Inputs
     .clk                               (clk),
     .rst_n                             (rst_n));

  /* fifo  AUTO_TEMPLATE (
    .din  (d),
  ); */

  fifo #
    (.DEPTH   (24),
     .T_DATA  (logic [7:0]))
  i_fifo
    (/*AUTOINST*/
     // Interfaces
     .din                               (d),                     // Templated
     .dout                              (dout),
     // Inputs
     .clk                               (clk),
     .rst_n                             (rst_n));

endmodule

// Local Variables:
// verilog-typedef-regexp: "^[tT]_"
// verilog-auto-inst-param-value:t
// End:
```

wsnyder, 2023-11-27:
Are you putting the Local Variables into both of the separated files?

cswmeta, 2023-11-27:
Local Variables were NOT in the separate module, only in the parent. After adding them to the separate module it now works. Thanks!

closed 2023-11-27 by wsnyder