veripool/verilog-mode #1851: typedefs in autooutput

amd013, 2024-01-03:
I am trying to get autoinst to connect outputs that are defined as a typedef.  For example, I have a package file that contains:

my_pkg.vh:

package my_pkg;

....
typedef struct packed {
logic  xyz;
some_other_type_t abc;
} my_type_t;

Then I have a file, my_module.sv with a module like:

module my_module
(
.....
output my_pkg::my_type_t my_sig,
output logic [my_pkg::my_param-1:0] my_vec,
....
);
....

Then I have another module that instantiates the above in file my_top_mod.sv

module my_top
(
/*AUTOINPUT*/
/*AUTOOUPUT*/
);
/*AUTOLOGIC*/

....
/*
my_module AUTO_TEMPLATE
(
...
);
// NOTE, the signal in question does not need renaming, so not part of the templates
*/

my_module u_my_module
(
.... // some hand coded sigs
/*AUTOINST*/);

....
// --
//------------------------------------------------------------------------------------------------
// Local Variables:
// eval:(verilog-read-defines)
// verilog-typedef-regexp:"_t$"
// verilog-library-directories: ("." "../../../path_to_pkg")
// verilog-library-extensions:(".v" ".sv" ".h" ".vh")
// verilog-auto-inst-param-value: t
// End:
//------------------------------------------------------------------------------------------------------

After running the AUTO's,

It adds the my_sig port connected to a signal called my_sig, but it does not declare my_sig in either the list of outputs or internal logics.  my_vec seems to work fine, but I am not sure if that is just because it is copying the indexes verbatim.

Note, this is a legacy design from previous generation of a design.  I saw it already has the autoinst syntax in it, so I did a "Kill-Auto's" and 
then "Recompute-Auto's" to see if I get the same results.  In previous design, it did add an output defined like this:

output logic [$bits(my_type_t)-1:0] my_sig,

But I suspect there was a post processing script run to add that, rather then autoinst.

wsnyder, 2024-01-04:
AUTOWIRE should be able to declare a type in a package, see for example autowire_pkg_bug195.v in the test cases.

Perhaps you don't have the verilog-typedef-regexp setting in every file?  If you think you do, please provide a standalone test case and I'll look at it.

Verilog-mode never makes "$bits" so that must have been something postprocessed on your side.

amd013, 2024-01-04:
Hi Thanks for the response.  Those didn't "seem" to help, but I have gotten it working.  However the first time it worked was after I closed emacs and reloaded it.  I made a test case, and made many changes.  I suspect the one that got it working was to make sure the package file was included, and then add eval:(verilog-read-includes).  However, since it worked after a reload, its possible that maybe other changes were the solution.  (note before this, in the module declaration, it added the port under "// Interfaces", rather than "// Outputs")

Thanks,
Mike

closed 2024-01-04 by wsnyder