veripool/verilog-mode #1854: autowire/logic on  Concatenation operators apostrophe

anythingelse0, 2024-01-08:
hi! I use  /*AUTOLOGIC*/ with '{}  on 
```
module test_port0
(/*AUTOARG*/
   // Outputs
   d
   );
    output [31:0] d[2];
   
endmodule // test_port

module test_port1
(/*AUTOARG*/
   // Inputs
   dd
   );
   input [31:0] dd[2]
endmodule // test_port1

module top()

/*AUTOLOGIC*/ // not work 

  /*
   test_port0  AUTO_TEMPLATE
   (
   .d\(.*\)         ('{d_1\1[],d_2\1[]}}),// 0 1 !!!!!!!!!! xxxx
   );
*/   
test_port0
u0(/*AUTOINST*/
   // Outputs
   .d					('{d_1[31:0],d_2[31:0]}})); // Templated)

  /*
   test_port1  AUTO_TEMPLATE
   (
   .d\(.*\)         ('{d_1\1[],d_2\1[]}}),// 0 1 !!!!!!!!!! xxxx
   );
*/      

test_port1
u1(/*AUTOINST*/
   // Inputs
   .dd					('{d_1d[31:0],d_2d[31:0]}})); // Templated

endmodule // top

   
```
i changed here add " **'?** " to match apostrophe structure
```
   ((string-match "^\\s-*'?{\\(.*\\)}\\s-*$" expr)
    (let ((mlst (split-string (match-string 1 expr) "[{},\']"))
          mstr)
      (while (setq mstr (pop mlst))
        (verilog-read-sub-decls-expr submoddecls par-values comment port mstr))))
```

wsnyder, 2024-01-08:
That makes sense, could you turn this into a pull request including this as a new test in tests/ and tests_ok?

wsnyder, 2024-01-11:
Thanks for fixing!

closed 2024-01-11 by wsnyder