veripool/verilog-mode #1856: "package_name::type_t" typed signals do not generate module ports

marcink, 2024-01-14:
In our project setup, a lot of ports use structures that are typedef'ed, with the declaration stored in a subsystem level package. 
 This leads to port declarations in modules looking like this:
```
input  subsystemPkg::structure_type_name_t   i_port_name,
output subsystemPkg::structure_type_name_t   o_port_name,
```

When instantiating modules using `AUTOINST`, these are all presence in the port connections under the section `// Interfaces`, although they are not interfaces, they are typed signals.

When I use `AUTOINPUT` and `AUTOOUTPUT` to declare module ports, these typed ports/signal do not generate a top level ports.  I tried `// verilog-typedef-regexp: "_t$"` but it did not help.

Is there a way to get auto-verilog to punch module input/output ports for such typed instance ports?  I would not expect any special handling, just copying the type-declaration of the port from instance.

marcink, 2024-01-14:
I tried using `/*AUTOINOUTMODULE("module_name", "port name regexp")*/`, and it generate:
```
subsystemPkg::structure_type_name_t   i_port_name,
subsystemPkg::structure_type_name_t   o_port_name,
```
without the required `input` or `output` keywords.

wsnyder, 2024-01-14:
This means Autos do not think it is a type.

Put
```
// Local Variables:
// verilog-typedef-regexp: "_t$"
// End:
```

at bottom of all files (submodule files too) and it will work.  If not, reopen and ]provide a complete example we can try.

closed 2024-01-14 by wsnyder

marcink, 2024-01-15:
Beautiful !!!

I was using verilog-typedef-regexp inly in current module, not in submodule.  Adding to submodules solved the issue. :)