veripool/verilog-mode #1858: Remove xemacs requirement to build verilog-mode?

Pinjontall94, 2024-01-20:
I'm trying to build the upstream verilog-mode, and removing the xemacs calls from the makefile does successfully build the package, but (despite the xemacs site's claims that all major linux distros have xemacs packages), I'm unable to build xemacs from source necessary to build *this* package from source. I understand wanting to keep xemacs compat for existing users, but considering [fedora dropped xemacs a while ago](https://fedoraproject.org/wiki/Changes/Deprecate_xemacs), I'm left wondering what to do here. Nix, Guix, and Homebrew also have no dedicated xemacs package so building from source is the only option left (and that doesn't appear to work either, even with xorg-x11-xbitmaps installed)

wsnyder, 2024-01-20:
Right now the rule is

`default : e/verilog-mode.elc x/verilog-mode.elc`

You can "make e/verilog-mode.elc" to build for just emacs.  Is that sufficient?  The target name isn't very pretty, I can add an alias if you want.

Pinjontall94, 2024-01-22:
Yeah, an alias might be a good idea, because otherwise I wouldn't know which to build! I understand not breaking backward compatibility tho ;/

closed 2024-01-23 by wsnyder