veripool/verilog-mode #1859: Port Coercion Issue on AUTOINPUT

ramyamohandoss, 2024-01-25:
Hi All, 

I want to be able to declare something as below

module A (
           input clk, 
           input rst_n,

           output logic valid,
           output logic ready,
           output logic [N-1:0] data
);

logic random_x;
logic random_y;

endmodule

Module with AUTO insertion would like 

module A (
           /*AUTOINPUT("_clk$")/
           /*AUTOINPUT("_rst_n$")/
           /*AUTOOUTPUT*/
);

/*AUTOLOGIC*/

endmodule

This expands to 

module A (
          
           input logic clk, 
           input logic rst_n,

           output logic valid,
           output logic ready,
           output logic [N-1:0] data
);

logic random_x;
logic random_y;

endmodule

How do I make input default to wire type while retaing output/logic for others? It seems like verilog-auto-wire-type: "wire"
, verilog-auto-declare-nettype: "wire" make all AUTOINOUT/OUTPUT/LOGIC wire type.

Thanks

wsnyder, 2024-01-25:
"logic" is a data type.  "wire" is not a data type, so there is no way presently to do this.

What is the ultimate problem with the code as is currently generated (e.g. what breaks)?

ramyamohandoss, 2024-01-25:
Verification Test Bench is trying to instantiate DUT where they are trying to drive stimulus through input. When input is expanded as "input logic" it cannot be driven and this requires input to always be default to wire nettype. This is called port coercion .

wsnyder, 2024-01-25:
Send me an edaplayground.com example showing the problem please.

ramyamohandoss, 2024-01-25:
Section 2.7 explains the problem that I am facing with port coercion. https://www.sutherland-hdl.com/papers/2007-SNUG-SanJose_gotcha_again_paper.pdf

When input is declared as "input logic" instead of "input"(default wire nettype), input port cannot be used to drive any values from Test Bench because net types allow, and resolve, multi-driver functionality. If AUTOINPUT nettype could be inidvidually controlled apart from AUTOOUTPUT/AUTOLOGIC, that would solve the issue.

wsnyder, 2024-01-25:
The paper describes why you don't want to allow port coercion (and should have the types), which I'm agreeing with, what I don't understand is why you DO want to allow coercion, and how you expect that to work with non-trivially typed ports (e.g. a "real").

ramyamohandoss, 2024-01-28:
https://edaplayground.com/x/76cH

Here is an example where input a cannot be driven or assigned from test bench since AUTO input expands to input logic and and not just input. 
In my example mentioned above clk and rst_n, cannot be driven from testbench because of the issue

wsnyder, 2024-01-29:
That example works fine on all the simulators I tried once I changed "design" to "designn".  Perhaps you aren't setting the SystemVerilog switch on the simulator?

closed 2025-02-28 by wsnyder