veripool/verilog-mode #1863: Word select in search stop at underscore

Diramu, 2024-02-08:
More of a How-To, than an issue.

I want cntl-s cntl-w   to  stop the word select at the next  '_' (underscore). (the behavour in vhdl-mode). 

What variable to I override?

Thanks.

Mark

wsnyder, 2024-02-08:
(modify-syntax-entry ?_ "." verilog-mode-syntax-table)

However it looks like many of the Verilog-Mode internals check where word breaks are, which this will break, so this is unlikely to work well.  Sorry.

closed 2024-02-08 by wsnyder

Diramu, 2024-02-08:
Well it does change the behavior to what I want, so thank you.  I have not seen any problems yet.

Diramu, 2024-02-12:
I've found a variable find-word-boundary-function-table  "This variable affects the behavior of forward-word and backward-word,"  I just need to just understand how to use it.