veripool/verilog-mode #1869: The regular expression .\(.*\) will result in an error if used alone.

1371906755, 2024-03-28:
for example:

/*test AUTO_TEMPLATE(
.\(.*\)  (\1[]),
);*/
test u_test(
/*AUTOINST*/);

wsnyder, 2024-03-28:
You have too many backslashes there.  You want:

```
  /* test AUTO_TEMPLATE (
    .\(.*\)     (\1[]),
    ); */
```

Full example here: https://github.com/veripool/verilog-mode/blob/master/tests/autoinst_dedefine.v

closed 2024-03-28 by wsnyder

1371906755, 2024-03-28:
for example:

/*test AUTO_TEMPLATE(
.\(\.*\)  (a_\1[]),
);*/
test u_test(
/*AUTOINST*/);

1371906755, 2024-03-28:
verilog_mode have a error

1371906755, 2024-03-28:
[image]
have a error

wsnyder, 2024-03-28:
Please attach a complete example as text (not screen shot) and say what the error is.

1371906755, 2024-03-28:
module example
#(
parameter ID_WIDTH     = 6
) 
(/*AUTOINPUT*/ 
/*AUTOOUTPUT*/ 
); 

/*test AUTO_TEMPLATE(
.clk            (clk),
.reset_n        (reset_n),
.\\(.*\\)         (alloc_\1[]),
);*/

test #(
/*AUTOINSTPARAM*/)u_test(
/*AUTOINST*/);

endmodule

module test
#(
parameter ID_WIDTH     = 6
) 
(
output[ID_WIDTH    -1:0] data_id,
input                   clk,             
input                   reset_n   
);
endmodule

1371906755, 2024-03-28:
This example will result in an error

wsnyder, 2024-03-28:
See my example in the answer above. You have too many backslashes.