veripool/verilog-mode #1870: This example(\\(.\*\\) ) will result in an error

1371906755, 2024-03-28:
module example
#(
parameter ID_WIDTH     = 6
) 
(/*AUTOINPUT*/ 
/*AUTOOUTPUT*/ 
); 

/*test AUTO_TEMPLATE(
.clk            (clk),
.reset_n        (reset_n),
.\\(.\*\\)         (alloc_\1[]),
);*/

test #(
/*AUTOINSTPARAM*/)u_test(
/*AUTOINST*/);

endmodule

module test
#(
parameter ID_WIDTH     = 6
) 
(
output[ID_WIDTH    -1:0] data_id,
input                   clk,             
input                   reset_n   
);
endmodule

wsnyder, 2024-03-28:
See #1869.

closed 2024-03-28 by wsnyder

1371906755, 2024-03-29:
"Sorry, for many regular expressions, a '' needs to be added to display them. Here's the actual code."

1371906755, 2024-03-29:
module example
#(
parameter ID_WIDTH     = 6
) 
(/*AUTOINPUT*/ 
/*AUTOOUTPUT*/ 
); 

/*test AUTO_TEMPLATE(
.clk            (clk),
.reset_n        (reset_n),
.\(.*\)         (alloc_\1[]),
);*/

test #(
/*AUTOINSTPARAM*/)u_test(
/*AUTOINST*/);

endmodule

module test
#(
parameter ID_WIDTH     = 6
) 
(
output[ID_WIDTH    -1:0] data_id,
input                   clk,             
input                   reset_n   
);
endmodule

1371906755, 2024-03-29:
I feel there is a discrepancy between my regular expression and reality. The photo is of the code I wrote
[image]

1371906755, 2024-03-29:
[image]

wsnyder, 2024-03-29:
No need to also send email.  You still have not indicated what error you are seeing.  With correct backslashes your example works fine for me.

```
module example
#(
parameter ID_WIDTH = 6
)
(/*AUTOINPUT*/
// Beginning of automatic inputs (from unused autoinst inputs)
input                   clk,                    // To u_test of test.v
input                   reset_n,                // To u_test of test.v
// End of automatics
/*AUTOOUTPUT*/
// Beginning of automatic outputs (from unused autoinst outputs)
output [ID_WIDTH-1:0]   data_id                // From u_test of test.v
// End of automatics
);
/*test AUTO_TEMPLATE(
.clk (clk),
.reset_n (reset_n),
.\(.*\) (\1[]),
);*/
test #(
/*AUTOINSTPARAM*/
       // Parameters
       .ID_WIDTH                        (ID_WIDTH))              // Templated
u_test(
/*AUTOINST*/
       // Outputs
       .data_id                         (data_id[ID_WIDTH-1:0]), // Templated
       // Inputs
       .clk                             (clk),                   // Templated
       .reset_n                         (reset_n));               // Templated
endmodule
module test
#(
parameter ID_WIDTH = 6
)
(
output[ID_WIDTH -1:0] data_id,
input clk,
input reset_n
);
endmodule
```

1371906755, 2024-03-29:
thank you ，maybe it's a version issue

1371906755, 2024-03-29:
I've updated to the latest version, but the error still persists. I've noticed the following phenomena

1371906755, 2024-03-29:
For the following piece of code：
module example
#(
parameter ID_WIDTH     = 6
) 
(/*AUTOINPUT*/ 
/*AUTOOUTPUT*/ 
); 

/*test AUTO_TEMPLATE(
.clk            (clk),
.reset_n        (reset_n),
.\(.*\)         (\1[]),
);*/

test #(
/*AUTOINSTPARAM*/)u_test(
/*AUTOINST*/);

endmodule

module test
#(
parameter ID_WIDTH     = 6
) 
(
output[ID_WIDTH    -1:0] data_id,
input                   clk,             
input                   reset_n   
);
endmodule

There will be the following error:
"example.v.emacsautotmp" [New] 43L, 947C [w]
:!emacs -batch -l /usr/share/emacs/site-lisp/verilog-mode.el example.v.emacsautotmp -f verilog-batch-auto
Loading /usr/share/emacs/site-lisp/site-start.d/desktop-entry-mode-init.el (source)...
Loading vc-git...
Processing /home/luoyi/apollo/MPU/rtl/submodules/bitmap/example.v.emacsautotmp
%Error: example.v.emacsautotmp:28: Can't locate `6' module definition
    Check the verilog-library-directories variable.
    I looked in (if not listed, doesn't exist):
        /home/luoyi/apollo/MPU/rtl/submodules/bitmap/example.v.emacsautotmp

shell returned 255
43 lines filtered
:!rm example.v.emacsautotmp
[No write since last change]

1371906755, 2024-03-29:
But I don't use /AUTOINSTPARAM/.

module example
#(
parameter ID_WIDTH     = 6
) 
(/*AUTOINPUT*/ 
/*AUTOOUTPUT*/ 
); 

/*test AUTO_TEMPLATE(
.clk            (clk),
.reset_n        (reset_n),
.\(.*\)         (\1[]),
);*/

test u_test(
/*AUTOINST*/);

endmodule

module test
#(
parameter ID_WIDTH     = 6
) 
(
output[ID_WIDTH    -1:0] data_id,
input                   clk,             
input                   reset_n   
);
endmodule

I can successfully generate code：

module example
#(
parameter ID_WIDTH     = 6
) 
(/*AUTOINPUT*/ 
// Beginning of automatic inputs (from unused autoinst inputs)
input                   clk,                    // To u_test of test.v
input                   reset_n,                // To u_test of test.v
// End of automatics
/*AUTOOUTPUT*/ 
// Beginning of automatic outputs (from unused autoinst outputs)
output [ID_WIDTH-1:0]   data_id         // From u_test of test.v
// End of automatics
); 

/*test AUTO_TEMPLATE(
.clk            (clk),
.reset_n        (reset_n),
.\(.*\)         (\1[]),
);*/

test u_test(
/*AUTOINST*/
            // Outputs
            .data_id                    (data_id[ID_WIDTH-1:0]), // Templated
            // Inputs
            .clk                        (clk),                   // Templated
            .reset_n                    (reset_n));              // Templated

endmodule

module test
#(
parameter ID_WIDTH     = 6
) 
(
output[ID_WIDTH    -1:0] data_id,
input                   clk,             
input                   reset_n   
);
endmodule

1371906755, 2024-03-29:
Alternatively, if I modify the code like this, it also generates code normally.

module example
#(
parameter ID_WIDTH     = 6
) 
(/*AUTOINPUT*/ 
/*AUTOOUTPUT*/ 
); 

/*test AUTO_TEMPLATE(
.clk            (clk),
.reset_n        (reset_n),
.\(.*\)d         (\1d),
);*/

test #(
/*AUTOINSTPARAM*/)u_test(
/*AUTOINST*/);

endmodule

module test
#(
parameter ID_WIDTH     = 6 
) 
(
output[ID_WIDTH    -1:0] data_id,
input                    clk,             
input                    reset_n   
);
endmodule

1371906755, 2024-03-29:
This is my screen recording.

https://github.com/veripool/verilog-mode/assets/26504317/b3f89f76-77cd-44bc-882b-a8c35c45c537

wsnyder, 2024-03-29:
You need a newline between the ) and instance name.

```
test #(
/*AUTOINSTPARAM*/)
u_test(
/*AUTOINST*/);
```

Not

```
test #(
/*AUTOINSTPARAM*/)u_test(
/*AUTOINST*/);

```

Sorry for the back-and-forth, I had reindented your modules when I imported from github which masked the problem for me.

BTW when posting to github use the "<>" icon which will put "```" around code blocks to make them look correct, which makes such problems including the first extra backslash one, less likely to cause confusion.