veripool/verilog-mode #1872: For .sv files, verilog-mode cannot be recognized in Vim

1371906755, 2024-03-31:
Hello, I'm a newly started engineer using verilog-mode. I found that when I use gvim with verilog-mode, it only recognizes .v files but not .sv files. Although I solved the problem by adding au BufRead,BufNewFile *.sv set filetype=verilog in .vimrc, it's not a good solution as it may cause other issues with .sv files. Do you have any other solutions?

[image]

https://github.com/veripool/verilog-mode/assets/26504317/74591d8d-e4ee-4836-8000-98de9589aaa3

wsnyder, 2024-03-31:
No idea, this is an Emacs tool not vim.

TerrenceSun, 2024-05-26:
Most likely it's your vim's filetype issue.
Type `:set ft?` to check it. 
The `verilog_emacsauto.vim` is only actived for `verilog` file type, not `systemverilog` file type.

closed 2024-05-26 by wsnyder