veripool/verilog-mode #1874: Emacs 29: Verilog Mode Font Hilighting: module keyword highlights following label in a comment block.

dspain55, 2024-04-01:
[image]

In Emacs 29 running under Cygwin (I sure hope that doesn't matter) the module keyword wants to hi-light the next word as a label even though they are both in a comment block. 

I tried to copy a screen capture of the issue. Haven't seen this before until I started using Emacs 29.
If the screen capture took okay, you can see the example for both type of comment blocks // and /* ... */. Doesn't happen on Emacs 27.1 under Ubuntu 22.04LTS. 

Hi Wilson. Sorry to "bug" you. More wonderful Emacs tomfoolery. Hope all is well with you and yours.

David Spain

wsnyder, 2024-04-02:
I recreated this, it appears a bug specifically with Emacs 29.  No verilog-mode change is needed; conformed works on 29.2 and 30.0.50; once Emacs 29.3 hits Cygwin, this will be fixed.

closed 2024-04-02 by wsnyder

dspain55, 2024-04-03:
Just to be thoroughly pedantic, I should have stated I see this in Emacs 29.3 under cygwin (see Emacs version line in the screen shot). I'll look for the fix in any subsequent version to 29.3. Thanks!