veripool/verilog-mode #1887: Accumulate bus width automatically

Nianuu, 2024-12-04:
Hi,
I would like to implement the following case.
Here are 2 modules:
module aaa(
    input [3:0] test_pin);
module bbb(
    input [2:0] test_pint);

and they are instanced inside this top module:

module top(
    input test_pin[6:0]);

    aaa aaa (
        .test_pin    (    test[3:0]    ));
    bbb bbb (
        .test_pin    (    test[6:4]    ));
endmodule

Both test pin are connected to the same net but different slices.
Is there any way to do this autumatically by using vl-width or other methods?

wsnyder, 2024-12-04:
Use [AUTOINSERTLISP](https://veripool.org/verilog-mode/help/#verilog-auto-insert-lisp) to set a counter to zero (and return "" to insert nothing).

Then make an [AUTO_TEMPLATE with lisp](https://veripool.org/verilog-mode/help/#verilog-auto-inst) that formats your pin name using that counter and vl-width (using number-to-string), and also increments that counter.

closed 2024-12-04 by wsnyder

Nianuu, 2024-12-06:
not understand.
Can u provide an example?
many thanks!

wsnyder, 2024-12-06:
```
module ExampInsertLisp;

   /*AUTOINSERTLISP(progn (setq my-cnt 0) "")*/

   /* x AUTO_TEMPLATE(
         .pin(@"(progn (setq my-cnt (+ my-cnt (string-to-number vl-width))) (number-to-string (- my-cnt 1)))":@"(number-to-string (- my-cnt (string-to\
-number vl-width)))"),
        );*/

   x x (/*AUTOINST*/
        // Inputs
        .pin                            (3:0));                   // Templated

   x x (/*AUTOINST*/
        // Inputs
        .pin                            (7:4));                   // Templated

   x x (/*AUTOINST*/
        // Inputs
        .pin                            (11:8));                  // Templated

endmodule

module x;
   input [3:0] pin;
endmodule

```