veripool/verilog-mode #1889: Logging debug information  of verilog-auto-input-ignore-regexp or verilog-auto-ouput-ignore-regexp to debug

engrvns, 2025-02-24:
Hi Verilog-mode team (@wsnyder ),

Users face a painful issue debugging why directives like -verilog-auto-input-ignore-regexp or -verilog-auto-output-ignore-regexp get ignored.

This issue can be complex, going beyond simple errors or wildcard issues.

Is there a way to log detailed mode information of these directives during Verilog expansion?

Thanks,
Fellow Emacs Verilog-mode user

wsnyder, 2025-02-24:
No, I'd suggest add `(message "...")` calls to the appropriate source code points of interest, e.g. at the place where that variable is used.

Closing since answered first question, but feel free to follow up.

engrvns, 2025-02-24:
Here' what's happening ...

I have an sub-module whose input port is driven by an internal logic signal.
The instance module port is also getting driven by an `AUTOINPUT` port leading to multi-driver logic.
I want to avoid the `AUTOINPUT` so I do this -

Code extract
```
// verilog-auto-input-ignore-regexp:""
// eval:(setq verilog-auto-input-ignore-regexp (concat
//      "\\("
          "\\|other_ports_to_ignore"
//        "\\|submodule_in1"
//      "\\)"))
```

This still creates submodule_in1  as top-level port.
I further declare submodule_in1 as a wire, but no matter what comes up as a top-level port.

The question is why is verilog-auto-input-ignore-regexp not working?
How can I debug this...?

wsnyder, 2025-02-24:
This works for me.  I'd suggest adding the `(message` to see what verilog-auto-input-ignore-regexp is as a value, and each string compared to it.

```
module ex_output_every (/*AUTOARG*/)

  /*AUTOINPUT*/

  sub sub
    (/*AUTOINST*/
     // Inputs
     .ssss                              (ssss));

endmodule

module sub;
   input ssss;
endmodule

// Local Variables:
// verilog-auto-input-ignore-regexp:""
// eval:(setq verilog-auto-input-ignore-regexp (concat
//      "\\("
//        "other_ports_to_ignore"
//        "\\|ssss"
//      "\\)"))
// End:
```

wsnyder, 2025-02-24:
Also make sure your eval's work, as by default they are turned off for security, see `enable-local-eval`

engrvns, 2025-02-25:
@wsnyder - Thank you for patiently working with me.
For the above code to work the below var should be true/ By default it is but If not 
It should be set as -
`(setq enable-local-eval t) 
`(sets enable-local-variables t)
`

About using message to debug it further it is to be done like this -
``(message "The value of variable is %s" verilog-auto-input-ignore-regexp)
``

wsnyder, 2025-02-25:
Yes, in your .emacs use
```
(setq enable-local-eval t) 
(sets enable-local-variables t)
```

And yes, add messages like that.

engrvns, 2025-02-25:
Another relevant article related to this thread is 
[](https://github.com/veripool/verilog-mode/issues/1398)

Closing issue

engrvns, 2025-02-25:
https://github.com/veripool/verilog-mode/issues/1398