veripool/verilog-mode #1890: very simple case with AUTOINST but getting invalid search error

unfrostedpoptart, 2025-02-28:
Hi.  I have a simple SV testbench and went to auto instantiate my DUT like this:

```
module_name #(/*AUTOINSTPARAM*/)
  i_module_name (/*AUTOINST*/);

...
// Local Variables:
// Verilog-library-directories: ("." "path-to-DUT-directory")
// End:
```

However, when I type c-c c-a, it populates the parameters but not the ports and generates this error:

`verilog-auto-inst: Invalid search bound (wrong side of point)`

What's going on?

Thanks!  - setup environment details below

```
Emacs  : GNU Emacs 29.1 (build 2, x86_64-pc-linux-gnu, GTK+ Version 2.24.32, cairo version 1.15.12)
 of 2023-12-08
Package: verilog-mode v2025-01-01-5f86652-vpo-GNU

current state:
==============
(setq
 verilog-active-low-regexp ".*_n"
 verilog-after-save-font-hook nil
 verilog-align-assign-expr nil
 verilog-align-comment-distance 1
 verilog-align-decl-expr-comments t
 verilog-align-ifelse nil
 verilog-align-typedef-regexp nil
 verilog-align-typedef-words nil
 verilog-assignment-delay ""
 verilog-auto-arg-sort nil
 verilog-auto-declare-nettype nil
 verilog-auto-delete-trailing-whitespace t
 verilog-auto-endcomments t
 verilog-auto-hook nil
 verilog-auto-ignore-concat nil
 verilog-auto-indent-on-newline t
 verilog-auto-inout-ignore-regexp nil
 verilog-auto-input-ignore-regexp nil
 verilog-auto-inst-column 60
 verilog-auto-inst-dot-name t
 verilog-auto-inst-interfaced-ports t
 verilog-auto-inst-param-value t
 verilog-auto-inst-sort nil
 verilog-auto-inst-template-numbers nil
 verilog-auto-inst-vector t
 verilog-auto-lineup 'declarations
 verilog-auto-newline nil
 verilog-auto-output-ignore-regexp nil
 verilog-auto-read-includes nil
 verilog-auto-reset-blocking-in-non t
 verilog-auto-reset-widths t
 verilog-auto-save-policy nil
 verilog-auto-sense-defines-constant nil
 verilog-auto-sense-include-inputs nil
 verilog-auto-star-expand t
 verilog-auto-star-save nil
 verilog-auto-template-warn-unused nil
 verilog-auto-tieoff-declaration "wire"
 verilog-auto-tieoff-ignore-regexp nil
 verilog-auto-unused-ignore-regexp nil
 verilog-auto-wire-type nil
 verilog-before-auto-hook nil
 verilog-before-delete-auto-hook nil
 verilog-before-getopt-flags-hook nil
 verilog-before-save-font-hook '((lambda nil (show-all)) whitespace-cleanup)
 verilog-cache-enabled t
 verilog-case-fold t
 verilog-case-indent 2
 verilog-cexp-indent 2
 verilog-compiler "echo 'No verilog-compiler set, see \"M-x describe-variable verilog-compiler\"'"
 verilog-coverage "echo 'No verilog-coverage set, see \"M-x describe-variable verilog-coverage\"'"
 verilog-delete-auto-hook nil
 verilog-fontify-variables t
 verilog-getopt-flags-hook nil
 verilog-highlight-grouping-keywords t
 verilog-highlight-includes t
 verilog-highlight-modules t
 verilog-highlight-translate-off nil
 verilog-indent-begin-after-if t
 verilog-indent-class-inside-pkg t
 verilog-indent-declaration-macros nil
 verilog-indent-ignore-multiline-defines t
 verilog-indent-ignore-regexp nil
 verilog-indent-level 3
 verilog-indent-level-behavioral 3
 verilog-indent-level-declaration 3
 verilog-indent-level-directive 3
 verilog-indent-level-module 3
 verilog-indent-lists t
 verilog-library-directories '("." "$USERWA/digital/src/rtl/dru")
 verilog-library-extensions '(".v" ".va" ".sv")
 verilog-library-files nil
 verilog-library-flags '("")
 verilog-linter "echo 'No verilog-linter set, see \"M-x describe-variable verilog-linter\"'"
 verilog-minimum-comment-distance 2
 verilog-mode-hook '((lambda nil (key-chord-define-local "pp" "parameter")) flyspell-prog-mode folding-mode untabify-mode)
 verilog-mode-release-emacs t
 verilog-mode-version "2025-01-01-5f86652-vpo-GNU"
 verilog-preprocessor "verilator -E __FLAGS__ __FILE__"
 verilog-simulator "echo 'No verilog-simulator set, see \"M-x describe-variable verilog-simulator\"'"
 verilog-tab-always-indent nil
 verilog-tab-to-comment nil
 verilog-typedef-regexp "_t$"
 verilog-warn-fatal nil
 )
```

unfrostedpoptart, 2025-02-28:
FYI - I just tried this in a different file and it worked fine.  I have no idea what's different.

wsnyder, 2025-02-28:
Try restarting emacs, and see if something was mis-cached.  Assuming not, there's some bad code (to Emacs' thinking at least) in your `...`, you can try commenting out nearly everything and bisect adding it back in to find it.

wsnyder, 2025-02-28:
Closing assuming you debug, but feel free to comment more.

closed 2025-02-28 by wsnyder