veripool/verilog-mode #1892: error trying to update Verilog-mode using list-packages

unfrostedpoptart, 2025-04-10:
Using emacs 29.1 on RHEL 8.9.  verilog-mode 2021.10.14.1273... built in.  Trying to update to 2025.1.1.100165 using list-packages. When I click on updated package, I get:

describe-package-1: Wrong type argument: char-or-string-p, ("Michael McNamara" . "mac@verilog.com")

What's wrong and how do I fix this?

unfrostedpoptart, 2025-04-11:
Here's the debug backtrace:

```
Debugger entered--Lisp error: (wrong-type-argument char-or-string-p ("Michael McNamara" . "mac@verilog.com"))
  package--print-email-button((("Michael McNamara" . "mac@verilog.com") ("Wilson Snyder" . "wsnyder@wsnyder.org")))
  describe-package-1(#s(package-desc :name verilog-mode :version (2025 1 1 100165202) :summary "major mode for editing verilog source in Emacs" :reqs nil :kind tar :archive "gnu" :dir nil :extras ((:url . "https://www.veripool.org") (:keywords "languages") (:maintainer ("Michael McNamara" . "mac@verilog.com") ("Wilson Snyder" . "wsnyder@wsnyder.org")) (:authors ("Michael McNamara" . "mac@verilog.com") ("Wilson Snyder" . "wsnyder@wsnyder.org")) (:commit . "78bc5949db489b143f7424540d6cc56fc529c9ea")) :signed nil))
  #f(compiled-function () #<bytecode 0x14e4ee3156d603af>)()
  help--window-setup("*Help*" #f(compiled-function () #<bytecode 0x14e4ee3156d603af>))
  describe-package(#s(package-desc :name verilog-mode :version (2025 1 1 100165202) :summary "major mode for editing verilog source in Emacs" :reqs nil :kind tar :archive "gnu" :dir nil :extras ((:url . "https://www.veripool.org") (:keywords "languages") (:maintainer ("Michael McNamara" . "mac@verilog.com") ("Wilson Snyder" . "wsnyder@wsnyder.org")) (:authors ("Michael McNamara" . "mac@verilog.com") ("Wilson Snyder" . "wsnyder@wsnyder.org")) (:commit . "78bc5949db489b143f7424540d6cc56fc529c9ea")) :signed nil))
  package-menu-describe-package(#<marker (moves after insertion) at 1137714 in *Packages*>)
  button-activate(#<marker (moves after insertion) at 1137714 in *Packages*> nil)
  push-button(1137714)
  funcall-interactively(push-button 1137714)
  command-execute(push-button)
```

wsnyder, 2025-04-11:
I don't generally use the package manager, and that redhat is fairly old so could be a package manager problem, please try the #1 option manual installation as described in the top readme https://github.com/veripool/verilog-mode

Closing but feel free to follow up here.

closed 2025-04-11 by wsnyder