veripool/verilog-mode #1901: Compile to C++ on macOS Sequoia 15.6.1

brunonanino, 2025-09-22:
Hello everybody,
My verilator comes from Brew:

`brew install verilator`

For testing, I tried to replicate the first  [example](https://veripool.org/guide/latest/example_cc.html) in the manual. Compilation fails because `cstdint` cannot be found.

An unsuccessful fix:

`verilator --cc -CFLAGS -stdlib=libstdc++ --exe --build -j 0 -Wall sim_main.cpp our.v`

Previously, I tried to install verilator from git following the [procedure](https://veripool.org/guide/latest/install.html) in the manual, but the execution of `./configure` terminated with 

`configure: error: g++ does not seem to successfully compile a simple C++ program`

although some lines above it said 

`checking that C++ compiler can compile simple program... yes`

At this point, I'm dubious whether verilator can seriously work on my mac

wsnyder, 2025-09-22:
Maybe the libc isn't installed?  I see a lot of different advice if google the error message, try some of what those suggest. If still doesn't work please post to the proper verilator github issues https://github.com/verilator/verilator/issues and a mac user may help, note this spot is for verilog-mode which is completely unrelated.

wsnyder, 2025-09-22:
Just to be clear the problem is the compiler is broken, it isn't really related to verilator at this point in configure stopping.

closed 2025-09-22 by wsnyder